DeviationTX/deviation #887: Throttle proportional timer

stylecheng, 2019-07-29:
When I set timer1:<cntdn-prop> and switch:<THR>, 
I want timer1 to be normal speed at 100% throttle，
the timer 1 is half the speed when the throttle is in the middle position，
and timer1 stops at -100% throttle.
But the actual situation is:
The timer1 stops when the throttle is at the midpoint.
Timer1 to be at normal speed at 100% throttle or 100% throttle.

Tygrys-1, 2019-08-07:
I solved this by defining a virtual channel (I named it THR0), defining it as a source: THR, linear curve, scale 50% and offset 50. It simply converts -100 to 100 to the 0 to 100 range.
Then it can be used as a timer switch or is can be shown on the main screen as a throttle value.

stylecheng, 2019-08-20:
Defining the virtual channel is a good idea. It's true that this problem can be solved. Thank you.

closed 2019-08-20 by stylecheng