DeviationTX/deviation #921: error of memory Jumper T8SG

golddogh, 2019-08-30:
HI ! 

I had the bug of trainning cable on my t8SG and now I can't reflash stm32. it is detected  and I can install but when I start the receiver it only displaying the USB logo. My pc detect a "defect" usb device.
In the advanced tab nothing is write, only "format" but in grey...

I think I have to format the memory or change the chip. do you have any solution for that ?

vladislavy, 2019-09-02:
Is it first generation (V1, black) T8SG?

golddogh, 2019-09-02:
It’s v2 plus, i had it in the x220hv pack

vladislavy, 2019-09-02:
Try to flash the custom build with bulk flash memory chip erase, start it and try to format transmitter USB drive and copy filesystem from firmware archive, then use Safely Remove Hardware. If ended successfully, flash any normal build.

https://www.deviationtx.com/forum/3-feedback-questions/8473-jumper-t8sg-v2-plus-bricked#74684

golddogh, 2019-09-02:
ok thanks I will try that.

it is possible to get the schematic of the jumper t8SG V2 plus ?  and the reference of the chip memory ?

golddogh, 2019-09-02:
it does not work :/ I think the chip must be replaced

closed 2019-09-04 by vladislavy