DeviationTX/deviation #925: "HAS_4IN1_FLASH" build option breaks DSM/DSMX protocol.

vladislavy, 2019-09-24:
It's impossible to bind with DSM/DSMX protocol. With disabled "HAS_4IN1_FLASH" all work fine.

vladislavy, 2019-09-26:
With BUILDTYPE_DEV I don't see any debug messages from SPISwitch_Init() -> detect(), but I can't find the reason.

joeclone, 2019-10-07:
Has there been any updates concerning this issue? Is such a waste that the latest features and protocols are not brought into 
https://www.deviationtx.com/forum/builds/8508-reference-nightly-build-deviation-v5-x-x#75174

vladislavy, 2019-10-08:
I can't find any reason why DSM/DSMX protocol failed with "HAS_4IN1_FLASH" enabled. Probably it's timing issue.

TheRealMoeder, 2019-10-08:
I also looked into it but cannot find the issue yet. Maybe @goebish has some idea?

joeclone, 2019-10-08:
Thanks a lot for taking the time to look inti this problem. In the mean time is there possibility of having a build that has working dsm/dsmx and the pxx protocols intact?

vladislavy, 2019-10-09:
Most likely it's timing issue. With "HAS_4IN1_FLASH" enabled, but "HAS_MULTIMOD_SUPPORT" disabled, DSM/DSMX protocol also work fine. I suspect multi register CYRF setup is too slow. Any additional check at PROTO_CS_HI and PROTO_CS_LO can slow down multi register setup. Problematic nrf24l01 protocol (I don't remember exactly) also can have the same problem.

vladislavy, 2019-10-15:
Why we use SPI2 clock 36/16=2.25MHz? DIY-Multiprotocol module use clock 36/8=4.5MHz

goebish, 2019-10-15:
that's not normal, we should use 4.5MHz as well.

vladislavy, 2019-10-16:
joeclone tested build with "HAS_4IN1_FLASH" and "HAS_MULTIMOD_SUPPORT" enabled and SPI2 clock 36/8=4.5MHz, but it doesn't help, he can't bind.

I can't test it, my devo8 can bind to Spektrum Rx at any case.

joeclone, 2019-10-16:
Meaning this problem is specific to devo10?

vladislavy, 2019-10-22:
With SPI2 clock 36/8=4.5MHz Hubsan (A7105) has dropouts, but it can be fixed the same way as in DIY-Multiprotocol, disable-enable interrupts inside PROTOSPI_read3wire(). Devo protocol (CYRF6936) work fine, about a hour of testing. I can't test CC2500 and nRF24L01.