DeviationTX/deviation #956: Flysky AllCell telemetry regression

stronnag, 2020-02-07:
Using T8SG V2 PLUS with FLysky FS-ia6B with either FS-CVT01 voltage sensor or iNav CC3D VBAT IBUS telemetry, the following is observed:

Published nightlies:
* 28ce50f : AllCell is displayed correctly
* 69a18e2 : AllCell is zero (RSSI still correctly received)

Latest:
* e90be75 : AllCell is zero (RSSI still correctly received)

On e90be75, reverting b5e8119454d0e3ad904001fd7342388f947e64e5 results in AllCell being displayed correctly again.

vladislavy, 2020-02-10:
Before b5e8119 :
SENSOR_CELL_VOLTAGE = 0x03
after :
SENSOR_EXTV = 0x03  - not used anywhere
SENSOR_CELL_VOLTAGE = 0x04

vladislavy, 2020-02-10:
You can try comment out both lines and add:
SENSOR_CELL_VOLTAGE   = 0x03,    // Avg Cell voltage

stronnag, 2020-02-10:
Thanks, I will try that.

stronnag, 2020-02-10:
That fixes it, thank you.

vladislavy, 2020-02-11:
I think SENSOR_EXTV (and FS-CVT01 sensor) has to be assigned to "Volt3". Can you say is "Volt3" box used for any sensor?

stronnag, 2020-02-11:
With  e90be75
* Main Page config, FS-CVT01 data is on TelemV1 (Volt1 / Volt3 does not appear in the Main Page Config list)
* Telemtry Config, FS-CVT01 data is on Volt1 (top row).

vladislavy, 2020-02-12:
Anyway FS-CVT01 data has not to be at AllCell, it's for cell voltage sensor. But, TelemV1 (Volt1) is for Rx voltage. I don't understand how FS-CVT01 data can be at two places, Volt1 and AllCell, it's for different sensors.

stronnag, 2020-02-12:
Now you have me thinking that the bench test bec is broken and passing VBAT through. Lucky the RX survived. In which case VBAT is nowhere to be seen in the latest nightly. To be confirmed later.

vladislavy, 2020-02-12:
You can try next nightly build. I route FS-CVT01 data to AllCell, since it's unlikely that two external voltage sensors will be used together.

stronnag, 2020-02-12:
I will do, but as far as I can see (having removed the bad bec), the only working solution (albeit for my TGS8V2+ / IA6B / (FS-CVT01 | iNav VBAT) appears to be reverting b5e8119. Unfortunately, due to the faulty BEC, https://github.com/DeviationTX/deviation/issues/956#issuecomment-584257915 is not valid.

vladislavy, 2020-02-12:
I added support for SENSOR_EXTV = 0x03, so I think all will work as before b5e8119.

stronnag, 2020-02-12:
I tested current head / 4700708 and there is no VBAT, just TX voltage:
[image]
whereas, reverting b5e8119 gives the expected:
[image]
front page, telemetry voltage alerts then working again.

vladislavy, 2020-02-13:
I have assigned SENSOR_EXTV = 0x03 (FS-CVT01) to AllCell. Scroll Telemetry monitor page, you have to see battery voltage lower, at Bat->line 5. If you use Cell1 for telemetry alerts, change it to AllCell. Cell1-Cell6 used for different sensor (SENSOR_CELL_VOLTAGE = 0x04).

stronnag, 2020-02-13:
Bat line 5 shows the RX (BEC) voltage, same as Bat line 1.

closed 2020-02-17 by vladislavy