DeviationTX/deviation #963: Firmware for T8SG v2 Plus Carbon Edition

150pilot, 2020-02-27:
I upgraded my firmware from original to nightly and now my LCD is mostly black. I think this is because the T8SG v2 Plus has an OLED screen but now the "Carbon Edition" has gone back to an LCD screen. But now I have no idea what firmware to flash to get my screen working again. Anyone?

eried, 2020-02-27:
The one that say _t8sg v2 plus_ in the name

150pilot, 2020-02-27:
That is what I upgraded to but my screen is now unreadable as my screen is an LCD and not an OKED like the T8SG v2 Plus. Is there a "Carbon Edition" firmware out there? I haven't been able to find it...

eried, 2020-02-27:
Oh I have the Carbon edition with a white OLED. I think someone in the forum commented about using an older fw for that case, but I do not know exactly which one

TheRealMoeder, 2020-02-27:
Try the t8sg_v2 without plus

150pilot, 2020-02-27:
Ok, thank you, I tried that and my screen works again but I'm not able to bind to my Eachine E010 now. Any ideas on that?

vladislavy, 2020-02-28:
What protocol do you use? "MJXq" or "E010"? If "E010", try "Freq-Fine" protocol option. Gradually change from 0 to -127 and to +127. When you can bind, change up and down up to lost connection, then calculate and set the middle value. Middle = (Down + Up)/2.

150pilot, 2020-02-29:
Yes, MJXq & E010. I will try this other procedure as soon as I can, thank you.

closed 2020-03-13 by vladislavy