DeviationTX/deviation #966: Parsing datalog.bin

albfan, 2020-03-02:
Using deviationtx on a jumper t8sg v2 plus I cannot parse any helpful info with datalog2csv.py

From this file:

[datalog.zip](https://github.com/DeviationTX/deviation/files/4272879/datalog.zip)

After looking around and fix things here and there I came out with:

```
#!/usr/bin/env python3

import os
import sys
from optparse import OptionParser

class Capture(object):
    def init(self):
        self.model = "None"
        self.rate  = "None"
        self.num_elem = 0
        self.max_elem = 0
        self.header_mask = []
        self.elem_names = []
        self.header_size = 3
        self.capture_size = 0
        self.data = []

    def __init__(self, data):
        self.init()
        header_mask_size = self.to_model(data[1])
        self.to_rate(data[2])
        print(header_mask_size)
        self.header_mask = data[3:3+header_mask_size]
        self.capture_size = self.parse_size(self.header_mask)
        self.header_size = header_mask_size + 3

    def add_elem(self, data):
        item = []
        idx = 0;
        for i in range(self.max_elem):
            if not (self.header_mask[int(i / 8)] & (1 << (i % 8))):
                continue
            size = self.get_size(i)
            val = self.format_data(i, data[idx:])
            print("type "+str(i))
            print(val)
            item.append(val)
            idx += size
        if idx != self.capture_size:
            print("Size mismatch")
            sys.exit(1)
        print('add item')
        print(len(item))
        print(item)
        self.data.append(item)

    def to_model(self, value):
        timers = ["Timer1", "Timer2", "Timer3", "Timer4"]
        telem_volt = ["Volt1", "Volt2", "Volt3"]
        telem_temp = ["Temp1(C)", "Temp2(C)", "Temp3(C)", "Temp4(C)"]
        telem_rpm  = ["RPM1", "RPM2"]
        telem_extra_items = 49
        telem_extra = []
        for i in range(telem_extra_items):
            telem_extra.append("TELEM_" + repr(i))
        inp    = []
        outch  = ["Channel1", "Channel2", "Channel3", "Channel4",
                  "Channel5", "Channel6", "Channel7", "Channel8",
                  "Channel9", "Channel10", "Channel11", "Channel12",
                  "Channel13", "Channel14", "Channel15", "Channel16"]
[... 305 lines cut ...]
82,100,96,-73,-100,-100,100,-100,-100,100,-100,-100,100,-100,-100,100
98,100,60,-87,-100,-100,100,-100,-100,100,-100,-100,100,-100,-100,100
-49,98,95,5,-100,-100,100,-100,-100,100,-100,-100,100,-100,-100,100
-100,100,90,96,-100,-100,100,-100,-100,100,-100,-100,100,-100,-100,100
-100,-31,-100,98,-100,-100,100,-100,-100,100,-100,-100,100,-100,-100,100
-22,-7,-98,-3,-100,-100,100,-100,-100,100,-100,-100,100,-100,-100,100
-2,1,-4,-2,-100,-100,100,-100,-100,100,-100,-100,100,-100,-100,100
-3,1,-4,-2,-100,-100,100,-100,-100,100,-100,-100,100,-100,-100,100
-3,1,-4,-2,-100,-100,100,-100,-100,100,-100,-100,100,-100,-100,100
-3,1,-4,-2,-100,-100,100,-100,-100,100,-100,-100,100,-100,-100,100
-3,1,-4,-2,-100,-100,100,-100,-100,100,-100,-100,100,-100,-100,100
-3,1,-4,-2,-100,-100,100,-100,-100,100,-100,-100,100,-100,-100,100
-3,1,-4,-2,-100,-100,100,-100,-100,100,-100,-100,100,-100,-100,100
-3,1,-4,-2,-100,-100,100,-100,-100,100,-100,-100,100,-100,-100,100
-3,1,-4,-2,-100,-100,100,-100,-100,100,-100,-100,100,-100,-100,100
-3,1,-4,-2,-100,-100,100,-100,-100,100,-100,-100,100,-100,-100,100
```

I'm just interested in AIL,ELE,THR,RUD, and I can take a look into `datalog.c` but if someone sees how to fix this from this input file it would be quicker and fixed for everyone.

The final goal is to log something as blackbox, for what a rate of 1s seems not enough, is there any limit to log this at 0,1s or so?

With that log would be great to create an overlay with sticks for fpv videos:

https://www.youtube.com/watch?v=NrJ4TcIT_h4

Tygrys-1, 2020-04-10:
If you are using the nightly after 1 mar 2019:

Find the code:
```
elif value == 0x7e:
            self.model = "Devo7e"
            inp = ["AIL", "ELE", "THR", "RUD", "HOLD0", "HOLD1", "FMODE0", "FMODE1"]
            rtc = ["Clock"]
        else:
```
and replace with

```
elif value == 0x7e:
            self.model = "Devo7e"
            inp = ["AIL", "ELE", "THR", "RUD", "HOLD0", "HOLD1", "FMODE0", "FMODE1"]
            rtc = ["Clock"]
elif value == 0x18:
            self.model = "Devo7e"
            inp = ["AIL","'ELE", "THR", "RUD", "AUX4", "AUX5", "HOLD0", "HOLD1", "FMODE0", "FMODE1", "SW_A0", "SW_A1", "SW_A2",
         "SW_B0", "SW_B1", "SW_B2", "SW_C0", "SW_C1", "SW_C2", "SW_D0", "SW_D1", "SW_D2","SW_E0", "SW_E1", "SW_F0", "SW_F1",
         "SW_G0", "SW_G1", "SW_H0", "SW_H1"]
            rtc = ["Clock"]
        else:
```
If you are using stock or eariler nightly: 

Replace the line:
            inp = ["AIL", "ELE", "THR", "RUD", "HOLD0", "HOLD1", "FMODE0", "FMODE1"]
    
with

    inp = ["AIL","'ELE", "THR", "RUD", "AUX4", "AUX5", "HOLD0", "HOLD1", "FMODE0", "FMODE1", "SW_A0", "SW_A1", "SW_A2",
         "SW_B0", "SW_B1", "SW_B2", "SW_C0", "SW_C1", "SW_C2", "SW_D0", "SW_D1", "SW_D2","SW_E0", "SW_E1", "SW_F0", "SW_F1",
         "SW_G0", "SW_G1", "SW_H0", "SW_H1"]

Than give it a try. 

You should maintain adequate tabs, as they are meaningfull in Python, simple copy-paste of above code might not work. 

As I am not a Python programmer I ported that script to Delphi and made GUI for it. These changes was backported back to Python. Not tested, use at your own risk ;-)

Tygrys-1, 2020-04-10:
Signature on your file show that you are using the older firmware, so use the second solution. Find 

elif value == 0x7e:

And replace the next inp= line according to previous pos.

albfan, 2020-04-10:
Will give it a try, and if works try to generalize the existing script and make a PR

albfan, 2020-04-12:
No changes with your suggestion. I  will make small test and see if results match my actions. If so I will do a MR.

Any initiative to record this as a blackbox log (data each millisecond or so?)