DeviationTX/deviation #986: Telemetry page for Bayang doesn't show any data on Devo7e

daniel-snowsurf, 2020-05-19:
There is only the page header but no labels and no data. Just one black rectangle in the middle of the screen. Checked nightly build of today 19/05/2020 and build myself master branch with same results. Built BRANCH_5.X to compare, and this one works fine. Compared both versions, and modified file pages/128x641x1/guiobj.h. In line 152 changed guiLabel_t box[10] to guiLabel_t box[36]. Bayang telemetry is now OK.
I have no clue on what the real value should be. Please, can somebody check that?

daniel-snowsurf, 2020-05-19:
The value seems to be the max number of labels and values of all telemetry pages. Without extended telemetry it should be 20 instead of 10. And with extended telemetry it's 36. I'll check it later.

TheRealMoeder, 2020-05-20:
Probably got broken in PR #880 when new telemetry values were added. Can you make a PR for the fix? If possible, also check color GUI in emulator.

daniel-snowsurf, 2020-05-23:
Already checked, with 20 items telemetry page shows properly, with 18 the bug shows again. Checked the emulator of the devof7, the telemetry page is fine with the wrong value and with the new one. Other Tx are not affected because they have extended telemetry. I'll try to do my first PR.

TheRealMoeder, 2020-05-24:
Thanks for your fix!

closed 2020-05-24 by TheRealMoeder