DeviationTX/deviation #989: Add support for Propel protocol

TheRealMoeder, 2020-05-20:
Working in MPM:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/Propel_nrf24l01.ino

Already preliminary work by @hexfet available, probably just some minor changes needed:
https://github.com/hexfet/deviation/tree/proto_propel

pascallanger, 2020-05-20:
Minor changes include:
 - bind phase completion
 - any ID and hoping freq
 - flags for the different features
 - telemetry for model status
 - ...
Only works with genuine nrf24l01. Some of the mpm modules do not have a genuine component like all the Jumper to date. I've tried to emulate it with another component but it didn't work due to the short timing...
Pascal

TheRealMoeder, 2020-05-20:
thanks for the heads up ;)