DeviationTX/deviation #991: Devo F7 Reset

Henry222222, 2020-06-12:
Ok, I installed The latest nightly build onto my Devo F7 and it appears to work as expected but when I go to change the protocol on the model menu it resets the remote and remove any settings applied.

chinderdelis, 2021-02-13:
mine reset too...solution?

whitedxi, 2021-08-18:
Same problem here. It seems the menu is waiting for somesort of feedback or reading faulty code. After that it crashes and restarts the RX.
Willing to help sending youtube vids if needed

whitedxi, 2021-08-18:
Seems the menu item is diffent aswell. the arrows look different

djtrance, 2022-07-06:
what version of devoF7 firmware are you ussing? 
do you have 4mb of memory or are you still ussing the stock memory?.

whitedxi, 2022-07-07:
Still stock memory

djtrance, 2022-07-07:
> Still stock memory Op do 7 jul. 2022 01:52 schreef Juan Andres ***@***.***>:
> […](#)

this can be the problem. the filesystem for devoF7 with stock memory are not stable. sometimes the devo has not enough memory to work and the firmware crash. im not working on the devoFs filesystem, I suggest you to update the stock memory for a 4mb memory, you will have the same functionality as devo10 and your computer can update the model configuration as a pendrive. if you do this, I can share my last devof7 code to you, and you can test it. Im working on some protocols and a new page for devo (OSD)

Anisotropic, 2025-11-06:
@djtrance Sorry for nekroposting. But is there info how to update the F7e memory? Is it enough to just change the memory chip itself? I have a good old Defo7E with changed STM chip and deviation firmware installed. Want to update to the newest possible version.