DeviationTX/deviation #993: LCD screen distorted

dronevate, 2020-07-09:
this is only my first post,  very new to the game, so go easy

i have proof that i read the manual and i have looked over several forums for information, did some diagnostic checking of my own and here is what i can provide:

i have a jumper t8sg plus carbon, i was trying to fly my first rc plane ever, successfully bound to a dsm/dsmx reciever, discovered i liked the idea of throttle being on aux 4 channel so i reassigned and re calibrated and when i went to fly my boot up screen was cut in half, as well as throttle check was scrambled,  main screen was fine, but when i went into the menus they are all scrambked as well.

it came loaded with deviation, dont know what version, havent gotten that far yet, just plugged it into the usb moments ago, about to figure out how to check firmware, pull error logs but before i did anything i thought maybe any info could help with debugging pics attatched
[image]
[image]

[image]
[image]

dronevate, 2020-07-09:
oh and reciever still binds to plane automatically and i have functionality on all channels i have assigned... just cant navigate any menus due to distortion, gonna say hardware is not the issue due to the screen having a crisp clean look on main page but who knows

MGaglia, 2023-04-29:
Hello Drovenate, I have the same issue. Have you solved it?