DeviationTX/deviation #1002: redpine white-screens on Devo7e

bkleiner, 2020-10-02:
With https://github.com/DeviationTX/deviation/pull/1001 merged the devo7e seems to white-screen upon repine protocol selection.
I do not own a 7e myself, only got reports by a buddy, to go off:

>  Flashed the latest nightly to Devo7e, get same result - when redpine is selected, I get a white screen and radio will not power down

What could be causing this issue? would be more then happy to submit another PR.

TheRealMoeder, 2020-10-03:
It crashes in `static void redpine_init(unsigned int format)`, probably not all statics are initialized in the modular build and this messes things up. Will debug further.

TheRealMoeder, 2020-10-03:
I tried to reproduce the issue, but everything works fine here with redpine in a modular build on t8sg_v2...

somewhatlurker, 2020-10-07:
I notice that the `format` variable is initialised to 0 and therefore placed in `.bss`. Does the module loader initialise bss to 0 like the buildchain expects? I don't see that in the code. If it doesn't, `format` may not be initialised before the first call to `redpine_init`, leading to an out-of-bounds read.

If the above is correct, we should probably zero the module memory before loading to ensure it can't happen again.

goebish, 2020-10-07:
unsigned format = 0; 
is bad for modular build, it should be initialized in redpine_init for example.

TheRealMoeder, 2020-10-10:
@bkleiner Next nightly version should contain the fix, please ask your buddy to try again with the updated build

bkleiner, 2020-10-10:
You just beat me to the punch, was just sitting down to give the bss theory a test.
We will test and report back, thanks!

bkleiner, 2020-10-13:
With the latest nightly selecting the redpine protocol no longer crashes the radio. However it seems to not bind up with the RX. There is a possibility my receiver implementation is at fault here, it seems to work just fine with other radios though.
We will investigate and report back.

My bud did mention that the TX Power setting only allows him to select "default" for redpine, not sure if that is related at all.

vladislavy, 2020-10-16:
Cneck if "hardware.ini" has
has_pa-cc2500   = 1

"default" will be with that line commented
; has_pa-cc2500   = 1

bkleiner, 2020-10-26:
Sorry for the delay. Seems there was a bit of a miscommunication on our end, with the latest fixes it’s all working.
Issue can be closed imho.

closed 2020-10-27 by vladislavy

TheRealMoeder, 2020-10-27:
Great! Thanks for helping to track the bug!