DeviationTX/deviation #1012: Bug ? T8SGV2 Plus - Protocol PXX - impossible to bind with PPM IN set to extend

flagadajones, 2021-07-27:
I tried to configure a head tracker from trainer Port.
but i can't bind my FrSky R9M Slim+ when PPM IN is different to none. 
Bind procedure doesn't start on T8SGV2.
I join my model file.

[model7.txt](https://github.com/DeviationTX/deviation/files/6886085/model7.txt)

gaetrua, 2021-07-30:
I think PXX protocol source need a modification for accept PPMIn mode for transmitter with JR Bay like T8SGxx
in the **pxxout.c** PXX not start if PPMIn is used. 
I understand it when transmitter hasn't Jr Bay and output PXX throught serial port, but in case of T8SG i think it will be possible to use PPMIn and PXX 
`
static void initialize(u8 bind) {

    CLOCK_StopTimer();
    if (PPMin_Mode())
        return;
`

flagadajones, 2021-07-30:
I think PXX protocol source need a modification for accept PPMIn mode for transmitter with JR Bay like T8SGxx
in the **pxxout.c** PXX not start if PPMIn is used.
I understand it when transmitter hasn't Jr Bay and output PXX throught serial port, but in case of T8SG i think it will be possible to use PPMIn and PXX
`
static void initialize(u8 bind) {

CLOCK_StopTimer();
if (PPMin_Mode())
    return;

hexfet, 2022-01-08:
The T8SG radios use the same serial port for the JR bay and the trainer port.  So PPMin cannot be used because it would interfere with telemetry data from the radio module.

closed 2022-01-08 by hexfet