DeviationTX/deviation #1014: Nightly builds broken

howels, 2021-08-31:
Nightly builds are failing and have been for some time.

howels, 2021-09-17:
Any updates on this? Nightly builds are still not available.

howels, 2021-09-19:
Have tried to build locally using the setup in https://github.com/DeviationTX/deviation/blob/master/.travis.yml and I get no errors. Anyone know why this CI build is failing?

gdenton61, 2021-09-26:
My local build (on Ubuntu 20.04.3) the devo7e build fails with the same error as shown on lines 1416 & 1417 of https://travis-ci.org/github/DeviationTX/deviation/builds/774621651:

arm-none-eabi-ld: objs/devo7e/frskyx_cc2500.elf section .bss' will not fit in region ram'
arm-none-eabi-ld: region `ram' overflowed by 18 bytes

So it looks like the latest code change to frskyx_cc2500.c is what broke the nightly builds?

howels, 2021-09-27:
https://github.com/DeviationTX/deviation/pull/1015 should fix the issue with flash-constrained targets. Please check it out, thanks.

Dropsleutel, 2021-11-07:
it seems still the nightlies which are daily provided are not updated.

howels, 2021-11-08:
> it seems still the nightlies which are daily provided are not updated.

Yep. No core maintainers are paying attention. This is a canary for me. I have bought an OpenTX/EdgeTX radio instead as I don't see enough active developers here in deviationTX.

The fix is posted in the PR above (https://github.com/DeviationTX/deviation/pull/1015), but we can see that it has been almost 5 months since the last successful nightly build (2021-06-11). Clearly there are tons of users but developers seem to have moved on.