DeviationTX/deviation #1017: CRSF device names longer than 12 characters break the device page

howels, 2021-10-03:
When using CRSF module protocol, if a device name is > 12 characters then a blank line is shown and the CRSF device page cannot be used.

I've tried to fix it myself but I cannot figure out where the 12 char limit is coming from and my attempts to pre-emptively cut the name strings short before they hit whatever is throwing the error but I'm getting nowhere. 

So I'm leaving this issue here in case someone who knows more about the GUI rendering process finds it.

hexfet, 2022-01-08:
What transmitter are you using?
What radio module are you using?
Can you please give an example of a device name that causes the failure?

I only have a Crossfire radio and it doesn't seem to have this problem.  The device name is "XF Transmitter" which is 14 characters and works correctly.

You can try reducing the constant CRSF_MAX_NAME_LEN to see if the behavior changes.

howels, 2022-01-09:
This happens with ExpressLRS devices. We can for instance see the RX which has a shorter name but the TX is not visible with the default name "BetaFPV Micro TX"
I experimented a lot trying to shorten the length of strings but nothing seemed to work, Deviation would show a blank line when the input string was too long - it would be strongly desirable to truncate the string instead of showing a blank line.

howels, 2022-01-09:
FYI I have tried a custom build with CRSF_MAX_NAME_LEN increased to 40 characters and the 14 character limit still applies.

hexfet, 2022-01-09:
Thanks for the pointer in the right direction.  Should be fixed by #1020.  Please give it a try.

howels, 2022-01-10:
> Thanks for the pointer in the right direction. Should be fixed by #1020. Please give it a try.

Thanks, I'll give it a go tomorrow. Really appreciate you looking at this!

closed 2022-01-18 by hexfet

howels, 2022-01-18:
BTW this is now fixed as of https://github.com/DeviationTX/deviation/commit/458a87e71dbaf7379d0bd94a44bc62243c6da258