DeviationTX/deviation #1028: CX10_NRF24L01 for Q222

j0e57, 2022-05-12:
Hi I need WLToys Q222 to be supported as Q242 option hard to bind.. 

I already see there is update on Multiprotocol code https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/931ba2bd68f345f1c6eb59f6f0bb64e25e484bc5/Multiprotocol/CX10_nrf24l01.ino
Im thinking the code can be updated / sync here but as it rather different development so if any experienced programmed can help sure faster than trying myself (I have some programming skill but not so low lvl like C)
TIA

BTW I can not join DeviationTX forum as no activation Email received

j0e57, 2022-05-15:
I already trying to update here https://github.com/j0e57/deviation/tree/j0e57-patch-1
but Im still can not test it as need higher windows version to compile, can anyone help to compile ? I try to test on my quad

hexfet, 2022-07-20:
Please try[ the build here.](https://www.deviationtx.com/downloads-new/category/2871-hexfet-j0e57-patch-1)  It is from your branch with one fix applied - case clauses don't create blocks.
```
diff --git a/src/protocol/cx10_nrf24l01.c b/src/protocol/cx10_nrf24l01.c
index 84330a24..2cbc348b 100644
--- a/src/protocol/cx10_nrf24l01.c
+++ b/src/protocol/cx10_nrf24l01.c
@@ -403,6 +403,7 @@ static void initialize_txid()
     case FORMAT_Q282:
     case FORMAT_Q242:
     case FORMAT_Q222:
+        {
                uint8_t offset=0;       //F_Q282
                      if(Model.proto_opts[PROTOOPTS_FORMAT] == FORMAT_Q242)
                             offset=2;
@@ -410,6 +411,7 @@ static void initialize_txid()
                                offset=3;
                      for(u8 i=0; i < 4; i++)
                                rf_chans[i] = 0x46 + 2*i + offset;
+        }
       
 //        rf_chans[0] = 0x46;
 //        rf_chans[1] = 0x48;

```
My recommendation for building on windows is to install VirtualBox, create an Ubuntu VM, and build on linux :)

j0e57, 2022-07-20:
Thanks will try & report asap

j0e57, 2022-07-23:
Seems Im missing something (seems forgot to add null on last array), my devo 10 keep restart while try to change CX10 submodel

j0e57, 2022-07-23:
Already fix on my branch https://github.com/j0e57/deviation/tree/j0e57-patch-1, including creating block on case clause.. Im still using Windows XP, even I edit github use phone as preview changes do nothing in my browser. Thank for help

hexfet, 2022-07-25:
You're welcome.  New build [4c28c51 is here](https://www.deviationtx.com/downloads-new/category/2875-hexfet-j0e57-patch-1).

j0e57, 2022-07-28:
Already test it, Menu system OK but still trouble binding.. will examine code again

j0e57, 2022-07-28:
Only found slight mistake on definition :
        uint8_t offset = 0;
while here should be like this:
        u8 offset =0;
do you think it make differences or error ? if not so maybe need further investigation :)

hexfet, 2022-07-29:
That doesn't make a difference.  Those two are equivalent.

I took a quick look and seems like your Q222 changes are equivalent to the ones in MultiModule.  Will take a closer look.

j0e57, 2022-07-29:
From https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module/page425#post36264456 from Dutch79 post seems my quad is Q222 v2 and maybe different binding methods, on my 1st post my original Tx is missing but when you reply my Tx already found so not much trouble for me but if my devo 10 can successfully bind sure good to anyone that interested :)

hexfet, 2022-07-30:
I took a closer look and don't see anything you missed.  Packet output looks fine in the emulator.  Maybe it's just that v2 is different.  If you want to make a pull request I'll merge your code - may help someone else.

j0e57, 2022-07-31:
ok, already create.. hope help opther

closed 2022-08-01 by j0e57