DeviationTX/deviation #1029: Traxxas protocol?

eried, 2022-07-05:
Could this protocol be added easily?
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/TRAXXAS_cyrf6936.ino

What is the status of deviationtx? I see the domain is dead.

djtrance, 2022-07-06:
i dont really know what it is the status of deviationtx, but i just returned to develop for it one week ago. i added the f949G protocol but i dint have the receiver to check it.  maybe i can port others protocols, but the question it is if someone it is working on this project.  i will still working on the devoF7 because i have it for long time as the same as devo10. and want to still flying with deviationtx. but would be good if other people can work on it again.  i dont have to much time but as i like deviationtx, i can try to add some others protocols. but it must be some protocols that lots of people can use it.

eried, 2022-07-08:
cool, I fixed few details in the audio messages stuff but I am completely lost on how to add a protocol. I got a traxxas rc car and it is kinda lame to use the default controller :D can you point me to your f949G file? do you have a tutorial or some overview?

djtrance, 2022-07-11:
sorry I don't have a tutorial for this.. actually my repo don't have the audio message...

hexfet, 2022-07-20:
I ported this protocol because I was supposed to be getting a Traxxas.  That fell through so I can't test.  Please try one of [these test builds](https://www.deviationtx.com/downloads-new/category/2870-hexfet-protocol-traxxas).

Unless you have a devo6.  Not enough space on that tx to add the new protocol.

eried, 2022-07-20:
awesome! thanks, I will test tomorrow. I have jumper t8sg v3

hexfet, 2022-07-21:
New [test build here](https://www.deviationtx.com/downloads-new/category/2872-hexfet-protocol-traxxas).  Had to add a couple things I missed.

eried, 2022-07-21:
Hey I get it to do *something* but erratically, and never binds https://photos.app.goo.gl/rKfqQLVC2oZGGNpa9 (at the end I turn the traxxas remote on to show that one is still binded). Anything I can help?

hexfet, 2022-07-22:
Can you confirm you tried version 3cd0c1d?

New version[ e86a5e4 is here.](https://www.deviationtx.com/downloads-new/category/2874-hexfet-protocol-traxxas)  The only change is that bind mode will wait forever because this protocol requires receiving a packet from the rx in order to bind.  It shouldn't matter whether bind is started first on the tx or rx.

djtrance, 2022-07-22:
@hexfet  do you now anything about a problem with the new lemo rx dsmp that cannot bind in deviation?. I actually test it and even that I received the telemetry from the rx, the flash red of binding never ends.  and the movement of the servo are not good. I tried with  2 new rx lemon.

hexfet, 2022-07-22:
@djtrance please don't post off-topic.  It's fine to start a new issue here or topic on the deviationtx forum.

First I've heard of DSMP.  According to the website it's "DSMX and DSM2 compatible with auto detection of both modes".  Auto detection of anything is at least a yellow flag.  Have you tried all the different combinations of tx settings for DSMX, DSM2, and the protocol options that are available?

Do you know if MultiModule gets the same results with DSMP?  I don't see any recent changes in that repo that seem related to DSMP.

eried, 2022-07-22:
> Can you confirm you tried version [3cd0c1d](https://github.com/DeviationTX/deviation/commit/3cd0c1d1938a5225740861397d79614a28b86236)?
>
> New version[ e86a5e4 is [...]

I tested -> https://www.deviationtx.com/downloads-new/category/2872-hexfet-protocol-traxxas (now I think it does not load)

I will test the new one tomorrow :) and give you feedback

hexfet, 2022-07-22:
If you tested that version then I expect the new version will just stay in bind mode forever.

There's a comment in the code "Works with Traxxas 6519 receivers".  I don't know anything about Traxxas receivers.  Is yours compatible with 6519?

eried, 2022-07-23:
Oh I dont know anything either :/ I got this traxxas summit 1:10 with everything stock. I can try to check which receiver it has, can you compile one with that flag just to test?

hexfet, 2022-07-25:
That's the build e86a5e4 above.  I don't know if the code comment means the protocol will only work with 6519 receivers, or if that was the receiver used to reverse engineer the protocol and other models are compatible.

djtrance, 2022-07-25:
> @djtrance please don't post off-topic. It's fine to start a new issue here or topic on the deviationtx forum.
>
> First I've heard of DSMP. According to the website it's "DSMX and DSM2 compatible with [...]

it is working now but I had to set up the channel manually and works