DeviationTX/deviation #1038: Support ELRS on T8SG V2 Plus

albfan, 2023-04-16:
From this closed PR https://github.com/DeviationTX/deviation/pull/1009, looks like at least from 4Jan 2022, ELRS (Express LRS) is supported out of the box.

Since last firmware availiable from T8SG V2 Plus are from 2021:

https://www.deviationtx.com/downloads-new/category/2830-2021-06-11

wonder if this feature is really supported but there's no firmware available.

So I build from sources the firmware using docker:

https://www.deviationtx.com/wiki/development/docker

***NOTE1***: I tried using podman but start do not do anything
***NOTE2***: I need to give correct permissions to volumen using :z

```
docker pull deviationtx/deviation-docker
mkdir devo_builds
docker create -it -v $PWD/devo_builds:/release:z --name deviation_build deviationtx/deviation-docker
docker start -it deviation_build
```

After succesfully build ( I disable default and emu_default, just enable t8sg_v2_plus)

I upload firmware on device.

As far as I know, ELRS needs to use CRSF (right) but nothing happens on modules.

I use R9M and R9MX (flashed with ELRS 3.2.0, with passhphrase, so not bind is needed):

[image]

but nothing happens on transmitter and receiver (transmitter blinks red and green) while receiver blinks red.

To confirm if I did something wrong, I flash a flysky fs-i6x with a modified opentx:

https://github.com/OpenI6X/opentx/releases/tag/openi6x-1.9.0

and after configure external module to CRSF, transmitter stays in green and receiver stays in green and red led.

Connecting to betaflight I can see receiver tab correctly update axis info:

[image]

So asking for help:

- Is CRSF the protocol to use (I read something about PXX, but probably for non ELRS versions)
- Should I build with some special flags to support ELRS?
- As I read OOTB support for ELRS 2.0, should I downgrade R9M from 3.2.0 to 2.x?
- I see only an option on t8sg CRSF options (a model1) should I add something else on usb disk of t8sg to support ELRS (like a lua script or similar). In this video https://www.youtube.com/watch?v=JSO6F_EDtoY the external module shows correctly an option for module attached (Happy model)

hexfet, 2023-04-23:
I replied to your post on the DeviationTX forum before looking here.  408f8c0 is the latest and does support CRSF and ELRS in particular.
A hardware mod is needed on the t8sg - details in the user manual protocols chapter.

However I think there may be an issue with the R9M.  Deviation outputs a non-inverted signal on the pin used for CRSF.  Since the R9M is from Frsky it probably expects an inverted signal.  Can you check what the s.port pin looks like with the fs-i6x?

albfan, 2023-04-24:
I tried your nightly build and looks the same as my local build. As commit is the same I just can tell by this menu on CSRF:

[image]

Anyway I install from your nightly https://www.deviationtx.com/downloads-new/category/2881-hexfet-nightly-build?download=47857:deviation-t8sg-v2-plus-v5-0-0-408f8c0-zip

But other videos I see around show the correct external module connected (with name not this model 1).

The hardware mod I think is related with connect t8sg pin1 with s.port. I did it temporarily this way:

[image]

but R9M keeps blinking red and green.

On frsky fs-i6x it stays green, and receiver stays green and red at same time.

Some details about the ports on R9M transmitter internals (I think GND, 5V and s.port are the same)

[image]
[image]

As said, I can build and make test (hardware and software) somehow I'm missing something but not sure what

hexfet, 2023-04-24:
I think it must be that the R9M expects an inverted signal.  The t8sg can only output a non-inverted signal.  On ESP32-based tx modules the ELRS configurator provides a checkbox for selecting inverted or non-inverted input.  That option is not available for the R9M.

albfan, 2023-04-24:
Wonder if I can put a hardware inverter

https://www.team-blacksheep.com/products/prod:qx7mod

It says 3.3V. is there any similar inverter with 5V?

Could the flag UART_INVERTED in express les solve that by software?

https://www.deviationtx.com/forum/protocol-development/8868-expresslrs#77217

albfan, 2023-04-25:
I tried a homemade inverter (with bc337 no transistor but doesn't work)

I found this hardware mod for r9m

https://www.rcgroups.com/forums/showpost.php?p=46673237&postcount=194

[image]

Can confirm It works.

@hexfet Is It expected some progress on CRSF menu? Now it only shows 3 empty rows and a 4th row with "model1"

albfan, 2023-04-25:
Just in case, a screenshot of t8sg s.port mod

[image]

hexfet, 2023-04-27:
The s.port mod looks fine.  I don't know about the R9M mod.  What do you mean by "it works"?

If no ELRS items appear (just the model name) then it means that Deviation is not receiving data from the module.  If you select the "model1" line you can change the baud rate to see if that makes any difference.  Do you have a logic analyzer or oscilloscope to look at the signal directly?

albfan, 2023-04-28:
> What do you mean by "it works"?

Looks the R9M mod left the signal uninverted, so t8sg correctly send axis and switches

> If no ELRS items appear (just the model name) then it means that Deviation is not receiving data from the module

Yes I have a logic analyzer, (saleae with pulseview) any specific samplerate and decoder to capture? if there's any guide or tutorial on forum didn't find it

hexfet, 2023-04-29:
So the control link works, but not the menu?  Are you receiving telemetry data on the telemetry test page?

I usually capture at 20MHz and use the serial decoder.  No tutorial but just need the one signal line (s.port pin) and ground.  Start the capture, turn on the t8sg, navigate to the protocol options.  That should be enough.

albfan, 2023-04-30:
No I don't even see telemetry.

Thanks for the tips, will create the capture and upload

albfan, 2026-02-20:
I give another try to this and was able to get r9m 2019 and r9mx binding and sending AETR (no other switch or dial even if I set 16 channels)

definitely I need the inverter mod on r9m, and I will be doing changes to confirm what is the last version supported (r9m do not support ELRS 4)

This time I use (for no reason) https://www.deviationtx.com/downloads-new/category/2902-hexfet-crsf-config-fix

but I can build from source if we can improve this, and I would love to sniff traffic if that helps in any way (I see 2.00M is not working for example)

I see in docs, a mod with trainer needs to be done to support telemetry, is that still the case?

https://www.deviationtx.com/user-manual/user-manual-7e-10

> 10.52. Protocol: CRSF (Crossfire)
The CRSF protocol sends Crossfire protocol serial data on the transmitter’s trainer port (tip connector). The trainer port ring is ground. On the T8SG PLUS transmitter the serial data also appears on the top pin in the JR module bay. To enable telemetry the serial input must be tied to serial output. For the trainer port tie tip to ring1. In the T8SG module bay tie the top and bottom pins together. Up to sixteen channels are supported.

I was able to bind every single time so is that only ffor nightly or should we change the manual:

> The CRSF bind and configuration operations are not yet supported. Use a PC to bind the Crossfire module and receiver before using with Deviation.

hexfet, 2026-02-21:
Should be set to half duplex in order to get telemetry and access configuration menus.  

The 2.00M rate is for use with TBS crossfire hardware.

We can close unless you have other questions.

albfan, 2026-02-22:
Do I need to set somethning on telemetry config after set full-duplex?

Let me confirm if r9m mod is needed and if last elrs firmware still works and I will close this