DeviationTX/deviation #1042: ELRS version 3.4.0 breaks backpack VTX channel selection

bnmoore, 2024-09-14:
When I go to change the VTX channel in the menu I can change the band, but not the channel. It is stuck at whatever the ELRS module was set to previously. This is only for ELRS versions 3.4.0 and above, 3.3.2 still works fine.
ELRS module: Betafpv 1W Micro TX
Radio: Jumper T8SG v2 plus

bnmoore, 2024-09-22:
I understand the issue now. Number selection GUI does not have a default step value so it doesn't let the selected value change. Text selection uses a step of 1. When I replace param->step with 1 in the code below, it works as expected. Tested on my radio and confirmed it changes the VTX channel.

src/pages/common/_crsfdevice_page.c
```
static const char *value_numsel(guiObject_t *obj, int dir, void *data)
{
    (void)obj;
    crsf_param_t *param = (crsf_param_t *)data;
    u8 changed = 0;

    param->value = (void *)(intptr_t)GUI_TextSelectHelper((intptr_t)param->value,
                                        param->min_value, param->max_value,
                                        dir, param->step, 10*param->step, &changed);
```

hexfet, 2024-12-07:
Thanks for the fix!  Merged #1043

closed 2024-12-07 by hexfet